grblHAL/Plugin_plasma #1: Plasma / THC with Atmel SAM3X8E (Arduino Due)

JohnR117, 2021-08-02:
Hi, I am a developer, I am working on a plasma right now. I intend to use grblHal with Atmel SAM3X8E (Arduino Due) as base hardware. I have doubts about how to add the Plasma / THC plugin to the main project? also if you are interested in testers and developers I like to be one . Thank you very much in advance.

terjeio, 2021-08-03:
Great, a bit of code has to be added to the driver to support the plugin. The plugin itself should be modified a bit to make use of the latest improvements made to auxillary I/O handling. Support for an analog input has to be added as well. I can make some of the changes needed.
I guess you will have to make a new board map or are you going to use one of the existing maps? And have you made the electronics needed for arc voltage translation to a level that the Due can handle?

JohnR117, 2021-08-03:
Thanks for the quick reply. 

**I guess you will have to make a new board map or are you going to use one of the existing maps?** 
Initially I am going to use a hardware mockup that uses the RAMPS 1.6 pin map and then I intend to develop my own hardware.

**And have you made the electronics needed for arc voltage translation to a level that the Due can handle?**
First I am going to use a voltage divider with a filter to couple the signal to the analog input of the Arduino Due. Then in the final stage of the project I will use this more professional hardware (https://aqtronic.com/product/iso107-plasma-voltage-opto-isolator-200a/) to do it.

Currently I am doing a review to understand the structure of the code of both the driver and the core of grblHAL, as well as the plugin. 

**Great, a bit of code has to be added to the driver to support the plugin. The plugin itself should be modified a bit to make use of the latest improvements made to auxillary I/O handling. Support for an analog input has to be added as well. I can make some of the changes needed.**
I would appreciate your help to achieve this. 

Finally, I have a question, the firmware (core) can send to the PC a real-time feedback of the position of the motors (even if it is with open loop control) for synchronization reasons.

terjeio, 2021-08-03:
> First I am going to use a voltage divider with a filter

Be sure to add overvoltage protection, a zener or a bunch of diodes in series? Check the datasheet for max analog input voltage (and max negative as well).

> I would appreciate your help to achieve this.

I can help with coding, but not much with testing. I plan to build a machine myself but I spend far too much time on grblHAL and ioSender so I do not know when I will be able to do so...

> ... real-time feedback ...

What do you define as real-time? Senders requests feedback by querying the controller - typically every 200ms. How often this can be done will mainly be limited by the transfer rate - if using an UART connection this is currently 115200 baud.

terjeio, 2021-08-10:
Due driver and plasma plugin has been updated and plasma plugin has been added to the project. I have modified the _ioports.c_ code to add an analog input (from Arduino analog pin 10). Currently this has been done in a rather crude way so I have not committed the change, here is the changed file: [ioports.zip](https://github.com/grblHAL/Plugin_plasma/files/6961432/ioports.zip)

To enable the plasma plugin add to _my_machine.h_
`#define PLASMA_ENABLE      1`
and uncomment 
`#define HAS_IOPORTS`
in _ramps1.6_map.h_.

Note that the code for inserting steps for moving the cutter up/down needs more work.

Here is a [link](http://linuxcnc.org/docs/devel/html/plasma/plasma-cnc-primer.html) to a bit more info for the LinuxCNC plasma implementation.

JohnR117, 2021-08-11:
Hi, Thank you very much for the update. I have not been able to communicate, because I am doing the hardware design and some other work stuff. As soon as I have something assembled and tested, I will upload it here

gxdeange, 2022-07-12:
Hi Team,

I am looking to integrate this plugin and was wondering if it is OK to use and if you could advise which Pin/Port has been assigned for the analog input? I had a look at ioports.c but cannot work out what has been assigned.

Regards
George

terjeio, 2022-07-13:
I do not have a plasma table to test with so I do not know if it is ok to use - testing needs to be done by someone who has one.
The _ioports.c_ provided above reads from analog port 10:

```
static int32_t wait_on_input (bool digital, uint8_t port, wait_mode_t wait_mode, float timeout)
{
    int32_t value = -1;

    if(digital && port < n_in)
        value = get_input(&aux_in[port], (settings.ioport.invert_in.mask << port) & 0x01, wait_mode, timeout);
    else if(port < n_analog_in) {
        value = analogRead(10);
    }

    return value;
}
```

gxdeange, 2022-07-13:
Thanks.

This is great information as the board I will develop will be used with
Plasma cutters and the THC capability will be a great feature.

I will look to incorporate PA10 as the analog input.

Cheers and have a great day
George

closed 2022-09-27 by terjeio