grblHAL/Plugin_plasma #2: How to include THC into the  Teensy 4.1 using web builder

midengc8, 2023-04-10:
Tried several time by adding the THC plugin on the web builder, but the firmware generated did not include the THC .
also like to know where all the control lines from the CNC Plasma cutter  connect to on the phil-barrett breakout board, can not find any documents on that. Thanks,

terjeio, 2023-04-11:
It fails to initialize if there is no analog input available - a message is output but lost due to the connection to the board happening after the message is sent. I guess I should add info in the $I response about failure to init...
BTW the analog input requirement could be removed if you have a THC controller that provides up/down digital signals - I need to modify the code for that.

@phil-barrett has made a board providing isolated analog input via I2C but this is not yet available? I have one and there is code in place to read data from it. The board is based on a MCP3221 ADC.

More work is needed on the plugin and I need someone with a machine to help out as I do not have one myself. You can do that?

Which aux pins to use for the control lines is configurable via $-settings, default configuration is available via the `$pins` command.

midengc8, 2023-04-11:
Thanks for the info. I am willing to test but I am new at this, my plasma cutter has the typical CNC connection, ie: torch start, ARC voltage, start cutting. I am using only the torch start control lines (connected to the spindle control line on the board) and I have been able to start and stop the torch leaving other CNC control lines open. The machine will go through the whole g code and cut to completion. I would like  to connect all the other control lines and implement the THC function. Any help will be greatly appreciated. 
On another note, how do you delete a macro that I created in ioSender? I have tried several combination, but fail to delete the macro.

Thanks.

terjeio, 2023-04-18:
ARC voltage requires an analog input - that would be the first hurdle. You can ask Phil if he has any available?

> On another note, how do you delete a macro that I created in ioSender?

Click _Edit_ in macros frame, select the macro, delete the contents and press _Ok_.

phil-barrett, 2023-04-18:
I may have a couple. Will be back from Vietnam on the 25th so will check then.

midengc8, 2023-04-23:
OK will wait for Phil to come back and go from there.

As for the macro, it was strange, I did  exactly    that before,  but  ioSender showed a blank macro, I can not get rid of it no matter how I tried.  I tried again after your info.  it disappeared, I must have done something wrong. Thanks.

svitol86, 2023-04-27:
Me and a friend are in the process of building a diy plasma cutting table pretty big and heavy (2.8m x 1.7m table) and made the first successful tests moving the x and y (2 motors, self squaring) axes using an Arduino mega with Grbl 1.1.
Now, before continuing with the build, we decided to use a dedicated board and are evaluating the one made by @phil-barrett, with the idea of integrating this plugin, even if it still in development. In fact i am willing to help with the development, if necessary.
From what I understand, a circuit is needed, with an ADC that reads the analog signal of the arc voltage and sends it to the board via I2C. Correct? Where can I find information and schematics about this circuit? Thanks

phil-barrett, 2023-04-27:
I checked and do not have any ADC boards left. Sorry - I only had 5 of them built. But, I am putting an order together for a batch of them (at least 10, maybe more) and will hopefully submit it tonight (jet lag willing).  I need to make a few changes (different V Reg and layout change for mounting holes for more secure mounting).  The factory was running 5-7 days for delivery in March so it shouldn't take long. 

I will sell these for a bit over cost (probably $10 USD) plus shipping/handling (around $5 USD for US delivery). Will be precise once I know actual costs.  Note, if I productize them, the cost will go up - early adopters get a break.  If I don't productize, I will release the files open source for anyone to use.

Also, I will create a github page for the design/board in a few days.

Phil

terjeio, 2023-04-28:
> In fact i am willing to help with the development, if necessary.

Great, the main challenge is to insert Z-axis motion - currently I use a separate timer to generate steps, somehow they should be added to the regular step generation.

> From what I understand, a circuit is needed, with an ADC that reads the analog signal of the arc voltage and sends it to the board via I2C. Correct?

It depends on the mode - an external THC that outputs up/down signals can also be used.
I have to modify the code a bit to allow plugin init when analog input is not available for that though.

phil-barrett, 2023-09-16:
After interminable delays, I have the ADC working with isolated voltage support. (had counterfeit isolator, that was one problem, design error another, sigh)  I'm trying to simulate the THC but don't get $352-$357 showing up in grbl settings?  Not sure how to proceed.

Also ioSender has these options for $350 - Off, N/A and Up/Down.  How does that map onto modes 0, 1 and 2 as listed in the THC project Readme?  Is 0 for Off?  Seems a bit counter intuitive.

[image]

I want to test the ADC to make sure grblHAL is seeing it.  I have a stand alone program that verifies the ADC is working correctly so I know the HW is ok. Suggestions?

terjeio, 2023-09-17:
> I'm trying to simulate the THC but don't get $352-$357 showing up in grbl settings? 

If the board is not powered or the ADC driver is not compiled in the settings will not show up.
If compiled in but not powered up this will be added to the `$pins `report:
`[PIN:MCP3221:0,Aux input 0,No power]`
If powered up:
`[PIN:MCP3221:0,Aux input 0,E0]`

`E0 `is the port number that can be used with `M66`, sending `M66E0L0` will trigger a read and the result will be added to the next real-tine report in the `|In:` element: `<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|F:0|Pn:PXYZ|In:3557>`
If NGC expression support is enabled the result from the last `M66` can also be retrieved by sending `$#=5399`, it is then reported back like this: `[PARAM:5399=3557.000]` 
Note that `M66` cannot be used when the plasma plugin has claimed the ADC port. If so the input from the ADC is shown in the _THC monitor_ flyout scaled and offset by the `$361` and `$362` settings. However the port cannot be claimed due to the ADC driver code beeing out of sync with core changes - I'll soon commit a fix for that.
And I'll have to add an ADC option to the Web Builder for including the ADC driver code - currently it is not available...

> Also ioSender has these options for $350 - Off, N/A and Up/Down. How does that map onto modes 0, 1 and 2 as listed in the THC project Readme? Is 0 for Off? Seems a bit counter intuitive.

It maps 1:1, for now. I have to update ioSender to remove the _N/A_ (Not Available) option when the ADC is not installed. The ioSender settings UI  generally hides the mapping between setting values and the textual representation, however it is still available via `$$`, e.g. `$$=350` will output this when the ADC is available and claimed:
```
    0 - Off
    1 - Voltage
    2 - Up/down
```
and this if not:
```
    0 - Off
    2 - Up/down
```

terjeio, 2023-09-17:
Fixes commited, Web Builder now adds the ADC driver code when the plasma plugin is selected - but it should be a separate option?

phil-barrett, 2023-09-17:
Thanks for the update.  will test today.

> separate option?

I don't have an opinion yet.  will play with it more and get back to you.

phil-barrett, 2023-09-17:
That works.  Really nice with ioSender 2.0.44p3.  A couple of nits - in the plasma flyout, the voltage display is in mV but only shows the highest 3 digits. Also, I think it is presuming the ref voltage is 4096 but in the case of the current ADC, it is 5V. I plugged in 1.2 for PAV scale ($361) and got exactly what my multimeter was showing. Sorry for reporting ioSender issues here. will do so in that repo.

I will look through the code but how does it go from input voltage to arc height?

I've still got a lot to work go through but have sent off for a small run of boards. That damn 3221 is relatively expensive for what it does. Looking at a cheaper TI ADC.

terjeio, 2023-09-17:
>  A couple of nits - in the plasma flyout, the voltage display is in mV but only shows the highest 3 digits

I'll have to create a new edge version, I'll do it tomorrow:

[image]

> I will look through the code but how does it go from input voltage to arc height?

$361 - $363 controls the conversion. I have yet to test this - and the feedback loop has to be verified and likely refined.
I have some new ideas about how to insert steps for the Z driver that I have to find time to work on, a test jig is waiting for me...

phil-barrett, 2023-09-17:
OK, that is in volts, not mV?  Makes sense.  Target voltage is 100V for most Plasma Cutters?

I'll look for 44p4.

phil-barrett, 2023-09-18:
44p4 looks good!  When I turn on the "spindle" I get this
[image]

Arc Voltage input is just via potentiometer divider.   Arc voltage scale is 0.05.  At 2.5V (actually 2.420) input, that gives 100.6 V.
[image]

Still working through this but good progress.

TorbanP, 2023-09-24:
I recently discovered this project while working on a custom board (and my own thc implementation) for fluidnc. Was actually modifying my torch ADC to add isolation and stumbled on @phil-barrett MCP3221 BoB. By the looks of it I should be moving on over here. 

I can help with THC testing but want to double check i have the right idea for setting up a testing environment as im new to grblHAL. Thinking a GRBLHAL2000 with the MCP3221 is safest place to start? I was thinking of whipping up a board that integrates the Isolated ADC with the USB-C (for I2C pendant) passthrough onto the two headers of mainboard. From what I can tell, the I2C is bog standard for pendant so I should be able to have both? 

I have two plasma tables that can be relatively quickly adapted to this project for testing. What's the best way for me to help?

phil-barrett, 2023-09-24:
Does the GRBLHAL2000 have the I2C header?  The pins on the [ADCI_1](https://github.com/phil-barrett/ADCI_1) are set up for the T41U5XBB so might need a bit of adapting.  I should be be getting the beta test batch on monday.

TorbanP, 2023-09-24:
From what i understand the grblhal2000 is using I2c for the breakout for the jog2k, which then fires i2c over a usb cable with some pricy extenders. Its a different header layout than what your using though. That said, your T41U5XBB looks better for testing anyhow so I ordered one. Thanks!

phil-barrett, 2023-09-26:
I anyone is interested, I just got the beta build of the ADC with vreg on board for isolated power input. I have a few available. It seems to work ok.  More about it [here](https://github.com/phil-barrett/ADCI_1/discussions/4).

terjeio, 2023-09-26:
Some new updates has just been committed.

[Here is a video](http://www.io-engineering.com/downloads/thc-simulation.mp4) of Voltage Mode in operation with my test jig.
At ~12 seconds THC disengages due to XY motion beeing completed - and that causes the stepper motor no longer track the potentiometer motion.

Currently the code only emit steps at 1KHz to keep the arc voltage in range, no PID and no acceleration and deceleration.
I have improved the step injection code somewhat and added `hal.get_micros()` to allow better control of the Z motor. Implementing PID control and acceleration/deceleration should thus be possible. @TorbanP - you can help out with that?

troth530, 2024-01-29:
@terjeio when I have no ADC connected I see this 
[PIN:MCP3221:0,Aux analog input 0,No power]
$$=350
    0 - Off
    1 - Voltage

Should $350 have 0 and 1 available with no ADC connected or  0 and 2 (up/down) as you mentioned above?  In web builder 0.8 a few days ago I could see the THC flyout in a version of IOsender I frankly cannot recall.  However I cannot see the flyout with web builder 0.9 IOsender (Edge 2.045p1) .  Can I assume this is expected or do I have another issue before I proceed with getting an ADC connected.

Also there are a lot of MCP3421 modules available that may help testing.  Would it be easy enough for a typical user to make the code adjustments to enable the 3421 from the 3221 code?

terjeio, 2024-01-29:
> Should $350 have 0 and 1 available with no ADC connected or 0 and 2 (up/down) as you mentioned above?

It depends on whether the ADC code is compiled in or not, not if it is connected. If compiled in the up/down ports are not claimed by default, change $368 and $369 to valid aux ports then reboot to add the option to $350,

>  Would it be easy enough for a typical user to make the code adjustments to enable the 3421 from the 3221 code?

Perhaps, if able to read the datasheet and configure the chip correctly. I would modify the 3221 code first to make it easier, no other changes are then required. If adding the code in a new file with new function names changes has to be made elsewhere.

troth530, 2024-02-02:
@terjeio the adc code is compiled in (#define MCP3221_ENABLE is set to 1 in my_machine.h). I have no $368 and $368 settings.  Is there a way to enable from the CPU_map file.  By the way I'm not sure I need up/down if arc voltage mode works.

For now I will make a breakout board for the 3221 adc, these are in stock at digikey.
Thank you

terjeio, 2024-02-02:
@troth530 You have to enable the [plasma plugin](https://github.com/grblHAL/iMXRT1062/blob/f940642b8ef9168fddc729f37d3a9016ece5bfbf/grblHAL_Teensy4/src/my_machine.h#L84). If you have done this already then what is your `$I` and `$pins` output. The settings will not show up unless there are enough aux input pins available.

TorbanP, 2024-02-29:
Finally getting back to this project. I have the mcp3221 running with a pot as per phils design, and ~2.5v driven in. I have compiled the firmware using the website and selecting the Plasma/THC plugin and do not see the THC monitor  menu on side. I also compiled from source selecting the THC module and the ADC module and am having same issue. also tried iosender edge 2.0.45p3. I see some discrepancy with above ([PIN:MCP3221:0,Aux analog input 0,Arc voltage], no E0) but can provide the following $I, $pins if this helps

I also see the above comment, but sure what enough aux input pins means in this context?
I know ill have to add some sort of ARC OK digital input as well at some point, is that the hangup?
```
$I
[VER:1.1f.20240228:]
[OPT:VNMHSL,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,ES,REBOOT,SED,RTC,ETH,SD]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:492K]
[DRIVER:iMXRT1062]
[DRIVER VERSION:240220]
[DRIVER OPTIONS:USB.1]
[BOARD:T41U5XBB]
[AUX IO:7,3,0,0]
[IP:]
[PLUGIN:PLASMA v0.12]
[PLUGIN:SDCARD v1.13]
[PLUGIN:FS macro plugin v0.07]
$pins
[PIN:16,Feed hold]
[PIN:17,Cycle start]
[PIN:14,Emergency stop]
[PIN:15,Probe]
[PIN:20,X limit min]
[PIN:21,Y limit min]
[PIN:22,Z limit min]
[PIN:36,Aux input 0,P0]
[PIN:30,Aux input 1,P1]
[PIN:34,Aux input 2,P2]
[PIN:35,Aux input 3,P3]
[PIN:41,Aux input 4,P4]
[PIN:28,Aux input 5,P5]
[PIN:29,Aux input 6,P6]
[PIN:MCP3221:0,Aux analog input 0,Arc voltage]
[PIN:2,X step]
[PIN:4,Y step]
[PIN:6,Z step]
[PIN:3,X dir]
[PIN:5,Y dir]
[PIN:7,Z dir]
[PIN:10,X enable]
[PIN:40,Y enable]
[PIN:39,Z enable]
[PIN:12,Spindle on]
[PIN:11,Spindle direction]
[PIN:13,Spindle PWM]
[PIN:18,Mist]
[PIN:19,Flood]
[PIN:31,Aux out 0,P0]
[PIN:32,Aux out 1,P1]
[PIN:33,Aux out 2,P2]
[PIN:24,I2C SCK]
[PIN:25,I2C SDA]
[PIN:0,RX,UART1]
[PIN:1,TX,UART1]
$$=350
    0 - Off
    1 - Voltage
    2 - Up/down
M66E0L0
error:39 - Value out of range.
```

terjeio, 2024-02-29:
> I see some discrepancy with above ([PIN:MCP3221:0,Aux analog input 0,Arc voltage], no E0)

When the analog input is "claimed" by the plasma plugin it can no longer be read by M66.

> I also see the above comment, but not sure what enough aux input pins means in this context?

Some driver/board combos does not have any/enough input pins, yours have.

> I know ill have to add some sort of ARC OK digital input as well at some point, is that the hangup?

Yes, you have to assign one of the aux inputs as arc ok with $367. See the[ board map](https://github.com/grblHAL/iMXRT1062/blob/5be5707e6f0c6708446d4155f0d1c3fe0f8beacf/grblHAL_Teensy4/src/T41U5XBB_map.h#L97-L109) for details. Input 6 (or 5) is perhaps the best choice since it has an optcoupler in front.

TorbanP, 2024-02-29:
Thanks, assigned input 6 to that a bit ago.  I still have no access to the other specific THC components $352-$357. Is there some option in web builder i'm missing beyond the required THC? I have tried quite a few variations of options. 

$$=352
error:53 - Setting is not available, possibly due to limited driver support.

terjeio, 2024-02-29:
Oops - I broke the settings handling earlier. I just committed a fix that hopefully works.

TorbanP, 2024-02-29:
That seemed to fix it, Thanks! Menu is visible and i can read the pot voltage