grblHAL/Plugin_plasma #3: THC Implementation guidance advice needed (:

petervanderwalt, 2023-04-27:
Hi! 

If we wanted to use a THC with the OpenBuilds BlackBox, what would your suggestion be? 

The THC we have an eye on can use Mode 2:
https://github.com/grblHAL/Plugin_plasma/blob/9a96784d7db9c0feb74141b760757e7e94fd9989/README.md?plain=1#L11

With the ports available on the BlackBox, we don't have three "spare" inputs for Arc OK, Up and Down. 
But - we might be able to cheat like we did with the parallel limits we did for the 4th axis with the plugin https://github.com/grblHAL/ESP32/blob/master/main/BlackBoxX32.c 

In this case we might be able to get away with the available ports if we: 

- Wire X, Y and Z switches all three to X input - use seperate homing sequences to home Z first, then X and then Y, using plugin similar to how we did the 4th axes / autosquare shared input to have them all share one input. 
- This clears up Y and Z limits input - the (optocoupled) port expects SIG closing to GND to activate - which works great with relay output THCs like the PromaSD and others - so we can use those two inputs for the UP/DOWN signals 
- The rarely used DOOR input can be used for Arc OK  (Or could we have the ability to bypass Arc OK for the cheaper machines and just use a G4 at your own risk if it fails to ignite - would accomodate the cheaper plasmas without a CNC port (Proma's voltage divider provides divided voltage sense to the THC, but no arc-ok) 
- Probe port is still available - so you can do material touch-off probing at the start of each cut to set initial heights (compensate for warp even before you can the THC, by touching off prior to piercing) 

Doesn't need an analog input as the THC handles the voltage sense side itself and just sends up/down and arc-ok (from the Inverter) 
So https://github.com/grblHAL/Plugin_plasma/issues/2#issuecomment-1502857866 might apply as well, we'd be forced to only support Mode 2 THCs due to our hardware - but we can document that clearly

terjeio, 2023-04-28:
A new map file and linked board code is the way forward - the plugin claims [auxillary inputs](https://github.com/grblHAL/Plugin_plasma/blob/9a96784d7db9c0feb74141b760757e7e94fd9989/thc.c#L544-L546) so these must be defined in the map.

petervanderwalt, 2023-04-28:
Awesome, thanks for the confirmation! Pushing forward on the hardware side of the project, just wanted to be sure we are on a track that could work (:

petervanderwalt, 2023-05-10:
Are there any options for SPI based port expansion built into grblHAL that we can use with the ESP32?  (Input and Output - if we need Inputs for the THC up/down/arcok)

terjeio, 2023-05-11:
> Are there any options for SPI based port expansion built into grblHAL 

No, but it could be added, e.g. by board specific code. Inputs would require an interrupt line triggered by the chip or continuoulsy reading by polling (DMA?)...

There is code support for I2C I/O expanders (currently output only), PCA9654E is supported by the ESP32 driver.

closed 2025-12-23 by terjeio