grblHAL/Plugin_plasma #4: Failure Following Simulated Arc On (Spindle on)

sntlewis, 2023-09-13:
OK, my hardware is not fully operational, so that may be the source of the issue.  Much of what I'm doing at this point amounts to simulation with relays and wires.  

I have a 5v relay connected to the spindle enable on the Teensy 4.1 T41U5XBB.  When I toggle spindle on ioSender, the relay closes (which is expected) and then opens a few seconds later.  This part of the behavior is similar whether the connection is USB or ethernet.

Now for a bit more.
1)  if connected via USB, ioSender (2.0.43) shows "arc on" in the status bar.  It doesn't change to "arc ok," "arc failed," or "arc delay."  I have the "arc ok" port set to [PIN:30,Aux input 1,Arc ok] per $pins which is correct.  I was thinking of simply hardwire AUX1 to 3v3 to simulate "high" (or setting to gnd for "low").  The plasma cutter "arc ok" is not much more than a NO circuit that is closed by Cutter.  Hence the idea of either jumper to 3v3 or GND.  This doesn't change the behavior.  Then ioSender will popup a message with "Unhandled exception occured: Object reference not set to an instance of an object" then ioSender closes.

2) if connected via ethernet (which will be my preferred connection), no "arc on" in status bar with similar msg from ioSender.

Am I "simulating" the "arc ok" incorrectly?  Is there a way to query the current status of the AUX digital pins from console?

I apologize in advance if this is not truly an issue.

terjeio, 2023-09-15:
> Then ioSender will popup a message with "Unhandled exception occured: Object reference not set to an instance of an object" then ioSender closes.

Is this with the [latest edge version](http://www.io-engineering.com/downloads) (2.0.44p2)?

If I compile with ethernet enabled the Teensy constantly reboots due to a memory leak. Arc ok check also fails when the arc ok input is inverted. Here is a corrected _thc.c_ for you to try with:

[thc.zip](https://github.com/grblHAL/Plugin_plasma/files/12616392/thc.zip)

And here is what it looks like in ioSender when I simulate arc ok (via  telnet connection):

[image]

> I apologize in advance if this is not truly an issue.

I need feedback to get this plugin ready so please report back all misbehaviours!

sntlewis, 2023-09-15:
1.	This was with the GitHub version ioSender XL version 2.0.43, so I tried the 2.0.44p2 version.  It did not make a difference either USB or Telnet.  I tried recompiling without ethernet support and connect USB.  No difference there, as I expected.
2.	Replaced the thc.c module and recompiled.  Noted that “console” output now shows “[PLUGIN:PLASMA v0.06]” indicating that the new thc.c module was used in the compile.  Using the 2.0.44p2 ioSender made no difference.  I did notice that now arc ok port moved to ST3 (I had my config at ST1), so rechecked with jumper of ST3 to 3v3 (right beside on the t41u5x5bb).  Again no difference.

 

Now what I did notice is:

 

What is different is I do not see a THC Monitor tab/window on ioSender(under either version).  Is there something I need to do/config setting in ioSender to activate/unhide the THC Monitor?

 

 

Stan

 

 <https://user-images.githubusercontent.com/20260062/268197189-c3bebb6d-7da6-4999-9363-60da5a133a58.png>

sntlewis, 2023-09-15:
> You are using the XL version? Seems like the edge build somehow failed for that, I have just uploaded new edge versions that should work.
>
> When the plugin has been successfully initialized THC should be in the new options string, this is what tells the sender to add the THC monitor flyout tab.
>
>  <https://user-images.githubusercontent.com/20260062/268286702-0349560b-7413-44cc-9448-9b0a5b54be08.png>
>
> If this is not there the what are your Aux ports settings?
>
>
> %
> ; grblHAL
> ; 1.1f.20230907
> ; [OPT:NMSL,35,1024,3,0]
> ; [NEWOPT:ENUMS,RT+,ES,REBOOT,SED,RTC,ETH]
> ; [FIRMWARE:grblHAL]
> ; [NVS STORAGE:*FLASH]
> ; [FREE MEMORY:492K]
> ; [DRIVER:iMXRT1062]
> ; [DRIVER VERSION:230903]
> ; [DRIVER OPTIONS:USB.2]
> ; [BOARD:T41U5XBB]
> ; [AUX IO:1,3,0,0]
> ; [IP:]
> ; [PLUGIN:PLASMA v0.06]
> ;
> [... 180 lines cut ...]
> ; 481 - Autoreport interval
> $481=0
> ; 484 - Unlock required after E-Stop
> $484=1
> %

Rather than include “instream” I have attached both a cut/paste of About info from ioSender 2.0.44p2 and my_machine.h

 

It doesn’t look like the “THC” option is appearing in the “newopts” line yet the line below appears with plasma v0.06.  Other than edit “my_machine” is there anything else to do to enable plasma?

 

On an off-hand, when I compile with options for telnet (both ethernet and telnet) in the my_machine.h, I still had to go into the grbl options via ioSender to enable telnet.

 

Maybe something is getting hung up in NVS?

 

Stan

terjeio, 2023-09-15:
Do you have the ADC board mounted?
Or are you compiling with it enabled and not present?
Asking since $350=1, this could be a reason for init failure. If you set it to 2 does `THC` appears as it should?

> Maybe something is getting hung up in NVS?

Could be, enabling/disabling plugins/options may lead to NVS allocation beeing changed and the single byte checksum does not always trap such changes. It is best to to reset driver/plugin settings on such changes by sending `$RST=&`

sntlewis, 2023-09-15:
Correct, I do not have the ADC board mounted.  I dug through Phil Barrett’s github to find the board/schematic.  I have ordered a few from Osh Park and parts from Mouser.  I hope that was the right board/parts.  I was going to continue to familiarize/simulate until those parts came in.

 

So, I am compiling with enabled (MCP3221) but not present.  $PINS returns expected msgs.

 

I tried $350=2 directly and through the Settings:GRBL panel via mode Up/Down.  No change.

 

I tried $RST=&, console returned message “[MSG:Restoring defaults]” as expected.  Retried $350=2.  No change.

 

Stan

terjeio, 2023-09-15:
Try this binary:

[grblHAL_Teensy4_Upload.ino.zip](https://github.com/grblHAL/Plugin_plasma/files/12622364/grblHAL_Teensy4_Upload.ino.zip)

> $PINS returns expected msgs.

Like this?

```
[PIN:36,Aux input 0,P0]
[PIN:30,Aux input 1,Cutter up]
[PIN:34,Aux input 2,Cutter down]
[PIN:35,Aux input 3,Arc ok]
```

sntlewis, 2023-09-15:
Yes.

 

Your binary shows this on the console (along with $pins output).  I did a $RST=& and powered the T41U5XBB down/up.  

 

[VER:1.1f.20230913:]

[OPT:NMSL,35,1024,3,0]

[AXS:3:XYZ]

[NEWOPT:ENUMS,RT+,ES,REBOOT,SED,RTC,THC]

[FIRMWARE:grblHAL]

[NVS STORAGE:*FLASH]

[FREE MEMORY:492K]

[DRIVER:iMXRT1062]

[DRIVER VERSION:230903]

[DRIVER OPTIONS:USB.2]

[BOARD:T41U5XBB]

[AUX IO:1,3,0,3]

[PLUGIN:PLASMA v0.06]

[PLUGIN:Servo v0.01]

[G54:0.000,0.000,0.000]

[G55:0.000,0.000,0.000]

[G56:0.000,0.000,0.000]

[G57:0.000,0.000,0.000]

[G58:0.000,0.000,0.000]

[G59:0.000,0.000,0.000]

[G59.1:0.000,0.000,0.000]

[G59.2:0.000,0.000,0.000]

[G59.3:0.000,0.000,0.000]

[G28:0.000,0.000,0.000]

[G30:0.000,0.000,0.000]

[G92:0.000,0.000,0.000]

[TLO:0.000,0.000,0.000]

[PRB:0.000,0.000,0.000:0]

$pins

[PIN:14,Emergency stop]

[PIN:16,Feed hold]

[PIN:17,Cycle start]

[PIN:15,Probe]

[PIN:20,X limit min]

[PIN:21,Y limit min]

[PIN:22,Z limit min]

[PIN:36,Aux input 0,P0]

[PIN:30,Aux input 1,Cutter up]

[PIN:34,Aux input 2,Cutter down]

[PIN:35,Aux input 3,Arc ok]

[PIN:31,Aux out 0,P0]

[PIN:32,Aux out 1,P1]

[PIN:33,Aux out 2,P2]

 

This shows on the “newopts” line that THC is enabled but still not showing the THC Monitor window.  It appears that what you did was “disable MCP3221” as well as “disable ethernet.”  I did the same thing on my end and recompiled (using PlatformIO but can switch to Arduino IDE 2).  Same results meaning no “THC Monitor” window/panel/Tab.

 

We’ll get this ironed out!

 

Stan

sntlewis, 2023-09-15:
Do I need to strap/jumper AUX(in) ports 1,2,3 high before starting ioSender?  Currently they are “floating” until I get the ioSender window open.

terjeio, 2023-09-15:
> Do I need to strap/jumper AUX(in) ports 1,2,3 high before starting ioSender?

No, there are on-board 47K pullups on those pins.

> I did the same thing on my end and recompiled (using PlatformIO but can switch to Arduino IDE 2).

And you got `THC` in newopts?
I believe the Arduino IDE _cannot_ be used to build grblHAL, at least not without modifications to the code.

> Same results meaning no “THC Monitor” window/panel/Tab.

Only the "tab" should be visible on startup, if it is not and you have `THC` in newopts and you are running 2.0.44p3 then there are more gremlins lurking.

sntlewis, 2023-09-15:
Ok, no jumpers.  I was able to use Arduino 2.2(iirc) to compile but much prefer PlatformIO as that is what I use for my 3D printer mainboards (BTT SKR 1.4).

 

Here is a partial of the $I console output..

[VER:1.1f.20230907:]

[OPT:NMSL,35,1024,3,0]

[AXS:3:XYZ]

[NEWOPT:ENUMS,RT+,ES,REBOOT,SED,RTC,THC]

 

And confirmed that I am running ioSender 2.0.44p2 NOT 2.0.44p3!  I get five tabs across the top and four tabs down the right side.  There is no tab anywhere that says “THC Monitor.”

 

Stan

terjeio, 2023-09-15:
> And confirmed that I am running ioSender 2.0.44p2 NOT 2.0.44p3!  I get five tabs across the top and four tabs down the right side.  There is no tab anywhere that says “THC Monitor.”

Update to [2.0.44p3](http://www.io-engineering.com/downloads) then. Working?

sntlewis, 2023-09-15:
Just went to your IO Engineering Website and saw the 2.0.44p3 version before I got your reply.  The short answer is THC Monitor is showing up both with USB and Ethernet connection.

 

It is working…

 

So I did a bit more simulation/testing…  

 

1.	Arc ok port is ST3 (AUX3).  Invert the port. Leave jumper off.  Enable spindle.  Console shows the proper number of retries then fail.  This is normal/expected behavior as I would expect.  Plasma cutter will have to close AUX3 to ground to activate “Arc OK.”
2.	Enabling MCP3221 and recompile breaks the THC Monitor!  This is true even if $350=2 is set afterwards!  This leads me to ask “If the MCP3221 dies, the only way to limp along in another mode is to recompile (without MCP3221 enabled)”?  I will circle back to this when I get the parts/board in.  
3.	Should the console show a msg if/when the MCP3221 is found?  I presume that the Arc Voltage on the THC will show the output of the MCP3221?  
4.	Is there a way to separate “breakaway” from “arc ok”? $350=0 or $350=1, Aux0?

 

All in all, I’d say it is working.

 

Stan

terjeio, 2023-10-02:
New version up with an attempt to enable PID feedback for positioning. $353 (P-gain) default value has been reduced to 1.0 (from 25.0) , as the previous default likely would lead to excessive ringing (violent moves).

> Enabling MCP3221 and recompile breaks the THC Monitor!  This is true even if $350=2 is set afterwards!

Do not enable the MCP3221 when it is not mounted.

> Should the console show a msg if/when the MCP3221 is found?

The `$pins` command will show the MPC3221 as not powered if it does not respond to an I2C probe.

> Is there a way to separate “breakaway” from “arc ok”? $350=0 or $350=1, Aux0?

I'll have to go back to the LinuxCNC documentation to find the definition of "breakaway" before I can answer, I am not used to the plasma cutter terminology.