grblHAL/Plugin_plasma #5: THC ADC resolution?

phil-barrett, 2023-09-30:
Sorry for putting this in issues - there is no discussion section.

The current ADC daughter board I make has 12 bit resolution.  Is 10 or 8 bit resolution acceptable? When I started out I assumed that 12 bits was needed but am not so sure.

Assuming a target Arc Voltage of 100V, 12 bit resolution means 24 mV per step, 10 - 96 mV/Step and 8 - 391 mV.  This assumes no noise so the reality is probably 3 to 4 times worse usable resolution, with an 8-bit ADC probably reaching a volt per step.  (disclaimer, I'm no analog guru).  

The reason I am asking is I've been looking at alternate ADCs.  The current MCP3221 is relatively expensive and seems to have limited/sporadic availability.  There is a 10 bit ADC - MCP3021 that looks like it is a drop in replacement for the MCP3221.  It appears to have broader availability and is a bit cheaper. The current driver in grblHAL should work with no changes. It yields a 12 bit value but the bottom 2 bits do not get set.  The datasheet does not define what they are set to but as long as it is consistent, it should not matter. I've got a couple on order and will try them out. The nice thing about the MCP3021, it has the same I2C address as the MCP3221 so no driver changes needed.

I've also been looking at several TI ADCs that have broader availability. They are well less expensive than the MCP3221 but require driver changes.  Probably small but still needs to be considered.

bLouChip, 2023-09-30:
@phil-barrett ;   So my experience with using a [Proma Compact THC SD](https://www.proma-elektronika.com/download/PDF/thcsd_en.pdf) is that it appears to have a 1V / 0.2mm sensitivity, that's 0.2mm increase or decrease of arc length during cutting.  You are correct that 100VDC is a fair nominal cutting arc voltage, range being 50 - 130VDC.  120 - 150VDC is a nominal  pilot arc/non-cutting voltage, and 150- 250DC is a fair range of open circuit voltage (not even a pilot arc), although this latter case would be a misfire or erroneous assembled torch tip. These are values are for blowback pilot arc power units, not High Freq pilot arc units which are not preferred in hobby/small business plasma tables due to higher than normal EMI transmission.

An ARC OK signal from the power source should negate the need to monitor/measure for anything other than cutting arc voltage. Regardless, a delay time control value in the sw to delay Z motion control after an arc is first detected can mask the need to pilot arc monitoring. Also, a max voltage deviation value from monitor voltage value could suffice as an indicator of cutting-to-pilot transition if the torch runs off the metal.  

So to answer your question, I believe 10 bit ADC could work in the case of the circuit having a bias of +40VDC or so, so the 100VDC monitor range is shifted to an actual arc voltage of 40VDC - 140VDC.

phil-barrett, 2023-09-30:
Interesting.  A 1V sensitivity tells me that 8 bit resolution would be fine, though 10 bit is probably to lowest I would go.  The current ADC design doesn't do an offset - probably isn't necessary with 12 bit.  Maybe not needed for 10 bit either.  

The current ADC design simply monitors a range of 0-200V so my numbers above are off by 2 with out any offset circuitry.  I think the 10 bit ADC still is ok though modifying the divider to have 150V (or 140) yield 5V would help.

terjeio, 2023-10-02:
Cheaper converters has less stable output in the lower bits than more expensive ones requiring averaging of samples to approach the specified resolution?

TorbanP, 2023-10-04:
Ive used the TI ads111x series for many projects, even through the part shortage they have been available. its also what I was using with my external THC attempt with fluidnc. jlcpcb can PCBA them as well.
[https://datasheet.lcsc.com/lcsc/1809192322_Texas-Instruments-ADS1115IDGSR_C37593.pdf](url)

terjeio, 2023-10-06:
Driver support for additional converters is something I'll look into, I am thinking about adding an additonal API level for ADCs and DACs since the low-level device code can be [pretty simple](https://github.com/grblHAL/iMXRT1062/blob/master/grblHAL_Teensy4/src/mcp3221.c).