grblHAL/Plugin_plasma #8: Plasma mode not available error

cdagneta, 2024-05-18:
Hey All

I'm running PLASMA v0.13 on 1.1f.20240404 in up/down mode.

When I fire the torch using the M4 command I get the warning "[MSG:Warning: Plasma mode not available!]" and the relay triggers, but none of the Plasma specific functionality appears to work. 

I'm sure I missed something somewhere but figured I'd reach out incase I haven't 😂

Plasma Config
```
; 350 - Plasma mode
$350=2
; 351 - Plasma THC delay
$351=1.0
; 358 - Plasma Arc fail timeout
$358=1.0
; 359 - Plasma Arc retry delay
$359=1.0
; 360 - Plasma Arc max retries
$360=3
; 366 - Arc voltage port
$366=0
; 367 - Arc ok port
$367=1
; 368 - Cutter down port
$368=3
; 369 - Cutter up port
$369=2
```

terjeio, 2024-05-19:
Which board?

cdagneta, 2024-05-19:
Teensy 4.1

```
{
  "driver": "iMXRT1062",
  "URL": "https://github.com/grblHAL/iMXRT1062",
  "board": "BOARD_T41U5XBB",
  "symbols": [
    "USB_SERIAL_CDC=1",
    "Y_AUTO_SQUARE=1",
    "PROBE_ENABLE=1",
    "KEYPAD_ENABLE=1",
    "BLUETOOTH_ENABLE=0",
    "PLASMA_ENABLE=1",
    "MCP3221_ENABLE=1",
    "NETWORKING_ENABLE=0",
    "ESTOP_ENABLE=1"
  ],
  "docker_instance": "",
  "grblHAL": "webBuilder"
}
```

terjeio, 2024-05-20:
There is an issue with the plugin claiming aux inputs - it does not do that successfully.

I wonder if there is a conflict with the keypad plugin, there should not be but can you try without it enabled?
FYI since I am away for a while I am not able to debug this now, I only have two development cards available and a Teensy is not among them. With one of them I have confirmed that pin claiming is working so the plugin itself should be ok.

cdagneta, 2024-05-20:
Removed the keypad plugin and I'm still getting the plasma mode not available warning.

cdagneta, 2024-05-20:
I've been poking this a bit today. I compiled the code from https://github.com/grblHAL/iMXRT1062.git and have been stepping through enabling the individual modules to figure out which part is breaking the pin assignment. It appears to be related to the MCP3211 plugin being included with the webbuilder by default when you choose the plasma module.  It does appear that the keypad plugin conflicts with the original auto claimed pins if I upload firmware with the keypad plugin enabled after testing with the plasma plugin, however re-configuring the pins to values that don't overlap with the keypad plugin resolves that issue after a reboot.

**Failed - With PLASMA, MCP3221, Y_AUTO_SQUARE**
```
[VER:1.1f.20240513:]
[OPT:VNMSL2,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,ES,REBOOT,SED,RTC]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:484K]
[DRIVER:iMXRT1062]
[DRIVER VERSION:240404]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[AUX IO:7,3,0,0]
[PLUGIN:PLASMA v0.13]
M3
[MSG:Warning: Plasma mode not available!]
$pins
[PIN:16,Feed hold]
[PIN:17,Cycle start]
[PIN:14,Emergency stop]
[PIN:15,Probe]
[PIN:20,X limit min]
[PIN:21,Y limit min]
[PIN:23,Y limit min 2]
[PIN:22,Z limit min]
[PIN:36,Aux in 0,P0]
[PIN:30,Aux in 1,P1]
[PIN:34,Aux in 2,P2]
[PIN:35,Aux in 3,P3]
[PIN:41,Aux in 4,P4]
[PIN:28,Aux in 5,P5]
[PIN:29,Aux in 6,P6]
[PIN:MCP3221:0,Aux analog in 0,No power]
[PIN:2,X step]
[PIN:4,Y step]
[PIN:8,Y2 step]
[PIN:6,Z step]
[PIN:3,X dir]
[PIN:5,Y dir]
[PIN:9,Y2 dir]
[PIN:7,Z dir]
[PIN:10,X enable]
[PIN:40,Y enable]
[PIN:38,Y enable]
[PIN:39,Z enable]
[PIN:12,Spindle on]
[PIN:11,Spindle direction]
[PIN:13,Spindle PWM]
[PIN:18,Mist]
[PIN:19,Flood]
[PIN:31,Aux out 0,P0]
[PIN:32,Aux out 1,P1]
[PIN:33,Aux out 2,P2]
[PIN:24,I2C SCK]
[PIN:25,I2C SDA]
[PIN:0,RX,UART1]
[PIN:1,TX,UART1]
$$
$0=10.0
$1=25
$2=0
$3=0
$4=7
$5=7
$6=1
$8=0
[... 369 lines cut ...]
$344=200.0
$345=200.0
$346=1
$347=5.0
$348=2.500
$349=25.000
$350=2
$351=3.0
$356=90
$358=3.0
$359=3.0
$360=3
$367=1
$368=3
$369=2
$370=224
$372=0
$384=0
$394=4.0
$398=100
$481=0
$484=1
$486=0
$673=1.0
```

terjeio, 2024-05-21:
> It appears to be related to the MCP3211 plugin being included with the webbuilder by default when you choose the plasma plugin. Compiling without the MCP3221 plugin works properly and adding the module in causes the warning to appear.

Good to know- I have to figure out why. The board I have here works with the MCP enabled, but that has the MCP code in `ioports_analog.c` unlike the Teensy where it is still in `ioports.c`. I'll move it when I am back home. I'll also update the webuilder to not enable the MCP driver by default and make it an option instead. 

> It does appear that the keypad plugin also mildly conflicts with the original auto claimed pins only if I upload firmware with the keypad plugin enabled AFTER testing with firmware that only has the plasma plugin first, the keypad plugin auto-claims pins that the plasma plugin had originally tried to auto-claim and causes a conflict.

This is/could be a tricky one to resolve, one option would be to not assign pins per default and force the user to do it instead. WIll have to think about it.

terjeio, 2024-06-25:
MCP3221 support (when available) must now be enabled separately in the [Web Builder](http://svn.io-engineering.com:8080) - in the _Optional inputs_ tab.

cdagneta, 2025-01-01:
> MCP3221 support (when available) must now be enabled separately in the [Web Builder](http://svn.io-engineering.com:8080) - in the _Optional inputs_ tab.

closed 2025-01-01 by cdagneta