grblHAL/Plugin_plasma #9: How to debug THC  setup on STM32F4?

wolfgangr, 2024-07-14:
I try to build a Plasma cutter with a STM32F411 "blackpill" .

It skd looks like working a bit, but does not really match the expectations from documentation and source code.
The pattern is similar to https://github.com/grblHAL/Plugin_plasma/issues/8, but not related to sth like MCP3211 : 
- Plasma Plugin is listed, but not in "NEWOPT"
- no THC section in ? output
- thc.c code is greyed out in STM32cubeIDE
- only part of the expected configuration variables are listed
- M6X commands throw error

I messed around with preprocessor conditionals, but in the end, after configring an analog input, I found that obviously at least some thc functionality is implemented.

Some details:
```
$I
[VER:1.1f.20240709:]
[OPT:VNMDSL+2,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,OS,ES,SED]
[FIRMWARE:grblHAL]
[SIGNALS:HSDTEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:31K]
[DRIVER:STM32F411]
[DRIVER VERSION:240624]
[BOARD:BlackPill THC]
[AUX IO:2,2,0,0]
[PLUGIN:Bootloader Entry v0.02]
[PLUGIN:PLASMA v0.13]
```
some blind dry run exercises (not all input is echoed)
```
?<Idle|MPos:456.000,456.000,0.000|Bf:100,1024|FS:0,0|Pn:T>
ok
error:2
ok
ok X500
?<Run|MPos:470.024,456.000,0.000|Bf:99,1024|FS:500,0|Pn:T|Ov:100,100,100|A:S>
ok
?<Run|MPos:490.552,456.000,0.000|Bf:99,1024|FS:500,0|Pn:T>
ok
?<Idle|MPos:500.000,456.000,0.000|Bf:100,1024|FS:0,0|Pn:T>
ok
$?<Idle|MPos:500.000,456.000,0.000|Bf:100,1024|FS:0,0|Pn:T|WCO:0.000,0.000,0.000>
$?<Idle|MPos:500.000,456.000,0.000|Bf:100,1024|FS:0,0|Pn:T>
```
Pinout:
```
[PIN:PB7,Feed hold]
[PIN:PB6,Cycle start]
[PIN:PA1,Safety door]
[PIN:PA0,Emergency stop]
[PIN:PB5,Probe]
[PIN:PB12,X limit min]
[PIN:PB13,Y limit min]
[PIN:PB15,Y limit min 2]
[PIN:PB14,Z limit min]
[PIN:PB9,Aux in 2,Stop disable]
[PIN:PB4,Aux in 3,P0]
[PIN:PB3,Aux in 4,P1]
[PIN:PB8,Aux analog in 0,Arc voltage]
[PIN:PA2,X step]
[PIN:PA4,Y step]
[PIN:PB0,Y2 step]
[PIN:PA6,Z step]
[PIN:PA3,X dir]
[PIN:PA5,Y dir]
[PIN:PB1,Y2 dir]
[PIN:PA7,Z dir]
[PIN:PC13,Steppers enable]
[PIN:PB2,Spindle on]
[PIN:PA8,Spindle PWM]
[PIN:PC14,Mist]
[PIN:PC15,Flood]
[PIN:PA15,Aux out 0,P0]
[PIN:PB10,Aux out 2,P1]
[PIN:PA10,RX,UART1]
[PIN:PA9,TX,UART1]
```
```
/* Pin Assignments:
 *
 *                             -----------
 *                         VB |           | +3V
 *       *   Step En/Dis  C13 |           | GND
 *  ?Mist? Coolant Flood  C14 |           | +5V
 *  ?Flood? Coolant Mist  C15 | *     - * | B9   I2C SDA - Aux in 2 -Stop disable
 *                        RST |      |K|  | B8   I2C SCL - Aux analog in 0 - Arc voltage
 *                 Reset   A0 |       -   | B7   Feed Hold *
 **Safety door/I2C strobe  A1 |           | B6   Cycle Start * 
 *       *        X Step   A2 |           | B5   Probe *
 *       *   X Direction   A3 |    / \    | B4   Encoder Index     - aux in 3, P0
 *       *        Y Step   A4 |   <MCU>   | B3   Encoder Pulse (B) - aux in 4, P1
 *       *   Y Direction   A5 |    \ /    | A15  Encoder Pulse (A) - aux out 0,P0
 *       *        Z Step   A6 |           | A12  USB D+ *
 *       *   Z Direction   A7 |   -   -   | A11  USB D- *
 *    Y2-step aka A Step   B0 |  |R| |B|  | A10  USART1_RX  ???
 *    Y2 dir A Direction   B1 |   -   -   | A9   USART1_TX  ???
 *     *  Spindle Enable   B2 |           | A8   Spindle PWM  *
 *  Aux out 2,P1 (sp-dir) B10 |           | B15  A Limit aka y limit min2
 *                        +3V |   -----   | B14  Z Limit *
 *                        GND |  |     |  | B13  Y Limit *
 *                        +5V |  | USB |  | B12  X Limit *
 *                             -----------
 */
```

upper end of configuration:
This is what worries me most - many variables listes in `thc.c` and in https://github.com/grblHAL/Plugin_plasma are missing
```
....
$132=200.000
$170=0.000
$171=0.000
$172=0.000
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=200.0
$346=1
$347=5.0
$348=2.500
$349=25.000
$350=1
$351=3.0
$358=3.0
$359=3.0
$360=3
$366=0
$367=255
$368=255
$369=255
$370=3
$372=0
$384=0
$392=4.0
$393=1.0
$398=100
$481=0
$484=1
$486=0
```

My gut feeling tells me that quite some stuff might still be broken...
Ho can I pin this down?

terjeio, 2024-07-14:
Set $367 to 0 or 1 to claim an aux input for the arc ok signal.

wolfgangr, 2024-07-17:
Ok, I'll try that in the next build.

Just busy designing a breakout, I encountered another possible problem:

**Is PB8 possible as analog-in?**
If I undestand the datasheet (STM32F411xC STM32F411xE; "Table 8. STM32F411xC/xE pin definitions" ) correctly, may be it's not?
ADC are assigned as follows:
```
PA0 ... PA7 is ADC1_0  ... ADC1_7
PB0 ... PB1 is ADC1_8  ... ADC1_9
PC0 ... PC5 is ADC1_10 ... ADC1_15
```

The MX-tool in STM32CubeIDE allow to select PB8 as analog in, and does not raise complaints in code generation.
What's the reliable source?
And, above all, what's implemented in grbl-HAL right now for STM32F411?

**Would it be advisable to route analog in to e.g. PA0 instead?**
just to be on the safer side of live...

terjeio, 2024-07-18:
> Is PB8 possible as analog-in?

Not according to the datasheet I have (xC variants), but it may work?.

> And, above all, what's implemented in grbl-HAL right now for STM32F411?

[These](https://github.com/grblHAL/STM32F4xx/blob/17440571911851c55f9aac2de8ff239200df3e04/Src/ioports_analog.c#L52-L83) inputs (only ADC1 for F411) .

> Would it be advisable to route analog in to e.g. PA0 instead?

I think so.

wolfgangr, 2024-07-19:
ah, good to know  before I klick "order" at JCLPCB....
Is it worth to dig into the code to enable more than one ADC?
I've put two analog input circuits on my current pcb draft. 
Is there any application at all in grbl yet for a second ADC?

terjeio, 2024-07-19:
> Is there any application at all in grbl yet for a second ADC?

Not that I am aware of except that is will be available for M66 and the value can be used in expressions (G-code).

wolfgangr, 2024-07-22:
Ok, that's worth a trial.
Had some space left on my board, so I kept a second anaolog input circuit there.