grblHAL/Plugin_plasma #10: THC plugin not working properly, showing false  "arc ok" among others

Toutoune86, 2024-09-19:
Hello,
I have tried  the THC ADC plugin from Phil Barrett. I had the web builder built the firmware accordingly and flashed it to my Teensy.
I then made a few tests, just with a potentiometer instead of real arc tension divider. I also made a few test with potentiometer signal unplugged, or even the ADC board unpowered. All results are the same. 
Whatever the circumstances : THC panel shows no "virtual red LED" lit  before first M3, then various LEDs lit :  THC up, THC enabled, Torch On, Arc OK, Breakaway. After M5, only the  THC up LED remains lit. 

It looks more like a soft issue than hardware, especially since I do not have the GRBL settings menu allowing the three $350 modes
[image]
only "off" and "Up/down" DId the web builder not work properly  ? What solutions could there be ?

Toutoune86, 2024-09-19:
Also, I just realize setting the plasma to off has the setting file changed to "$350=0" and switched to "Up/down" has it writ in the settings "$350=2", which is not understandable to me, not consistent with the [operation mode description](https://github.com/grblHAL/Plugin_plasma).

terjeio, 2024-09-19:
Is the ADC board recognized? If it is not powered it will not show up in the `$pins` command output as such.
If it is and has not been previously it may help to reset plugin settings to default values by sending `$rst=&`, this should enable the third $350 option.
FYI $366-$369 controls the THC port bindings, they are available under the _Aux ports_ settings group.

DBS3262, 2024-10-23:
I'm currently having the same problem. I have 3.2VDC between the 3V3 and GND output terminals on the ADC. When I run the $pins report, PIN:MCP3221 is not visible. I've attempted the above reset and have not had any luck. When spindle is turned on, the slide comes to life, but without the voltage input, the THC starts to lower the torch. Here is the current config and pins report:
$i
[VER:1.1f.20241019:]
[OPT:VNMZTSL,100,1024,5,0]
[AXS:5:XYZAB]
[NEWOPT:ENUMS,RT+,HOME,ES,REBOOT,SED,RTC,ETH,THC,SD]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:484K]
[DRIVER:iMXRT1062]
[DRIVER VERSION:240928]
[BOARD:E5XMCS_T41]
[AUX IO:3,3,0,3]
[IP:]
[PLUGIN:PLASMA v0.13]
[PLUGIN:SDCARD v1.13]
[PLUGIN:FS macro plugin v0.09]
$pins
[PIN:16,Feed hold]
[PIN:17,Cycle start]
[PIN:14,Emergency stop]
[PIN:15,Probe]
[PIN:20,X limit min]
[PIN:21,Y limit min]
[PIN:22,Z limit min]
[PIN:23,A limit min]
[PIN:28,B limit min]
[PIN:36,Aux in 0,P0]
[PIN:30,Aux in 1,P1]
[PIN:34,Aux in 2,P2]
[PIN:35,Aux in 3,Cutter up]
[PIN:41,Aux in 4,Cutter down]
[PIN:29,Aux in 5,Arc ok]
[PIN:2,X step]
[PIN:4,Y step]
[PIN:6,Z step]
[PIN:8,A step]
[PIN:26,B step]
[PIN:3,X dir]
[PIN:5,Y dir]
[PIN:7,Z dir]
[PIN:9,A dir]
[PIN:27,B dir]
[PIN:10,X enable]
[PIN:40,Y enable]
[PIN:39,Z enable]
[PIN:38,A enable]
[PIN:37,B enable]
[PIN:12,Spindle on]
[PIN:11,Spindle direction]
[PIN:13,Spindle PWM]
[PIN:18,Mist]
[PIN:19,Flood]
[PIN:31,Aux out 0,P0]
[PIN:32,Aux out 1,P1]
[PIN:33,Aux out 2,P2]
[PIN:0,RX,Primary UART]
[PIN:1,TX,Primary UART]

I'm new to this , so I apologize in advance if this an obvious fix.

terjeio, 2024-10-24:
@DBS3262 Have you compiled with the MCP3221 option enabled?

[image]

DBS3262, 2024-10-24:
That was the original issue. Thanks Phil for the help with that!
(In the interest of full disclosure, I may need the correct pullup resistors. I tried reaching out to manufacturer but didn't get a fruitful response. I've since added a pullup circuit to my bread board and was able to get both SC and SD pulled high, but it hasn't changed anything.)
I now have the MCP3221 ADC, but not successfully reading voltage. Since the re-flash, I no longer have the plasma fly out. I did gain the voltage mode setting, but failed to gain any of the other settings such as scale or offset. I've tried resetting via $RST=&, and I've tried turning on the spindle to get thing to come to life but without success.  Turning on the spindle gets the following error: [MSG:Warning: Plasma mode not available!] I have also attempted to change the mode without any observable changes, and commanding a reset while in one of the other modes returns me to voltage mode.  Lastly M66E0L0 gets error 39 "Value out of range." Here is the current config and pins report:
$i
[VER:1.1f.20241023:]
[OPT:VNMZTSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,ES,REBOOT,SED,RTC,ETH,SD]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:484K]
[DRIVER:iMXRT1062]
[DRIVER VERSION:240928]
[BOARD:E5XMCS_T41]
[AUX IO:6,3,0,0]
[IP:]
[PLUGIN:PLASMA v0.13]
[PLUGIN:SDCARD v1.14]
[PLUGIN:FS macro plugin v0.10]
$pins
[PIN:16,Feed hold]
[PIN:17,Cycle start]
[PIN:14,Emergency stop]
[PIN:15,Probe]
[PIN:20,X limit min]
[PIN:21,Y limit min]
[PIN:22,Z limit min]
[PIN:36,Aux in 0,P0]
[PIN:30,Aux in 1,P1]
[PIN:34,Aux in 2,P2]
[PIN:35,Aux in 3,P3]
[PIN:41,Aux in 4,P4]
[PIN:29,Aux in 5,P5]
[PIN:MCP3221:0,Aux analog in 0,No power]
[PIN:2,X step]
[PIN:4,Y step]
[PIN:6,Z step]
[PIN:3,X dir]
[PIN:5,Y dir]
[PIN:7,Z dir]
[PIN:10,X enable]
[PIN:40,Y enable]
[PIN:39,Z enable]
[PIN:12,Spindle on]
[PIN:11,Spindle direction]
[PIN:13,Spindle PWM]
[PIN:18,Mist]
[PIN:19,Flood]
[PIN:31,Aux out 0,P0]
[PIN:32,Aux out 1,P1]
[PIN:33,Aux out 2,P2]
[PIN:24,I2C SCK]
[PIN:25,I2C SDA]
[PIN:0,RX,Primary UART]
[PIN:1,TX,Primary UART]

terjeio, 2024-10-25:
> [PIN:MCP3221:0,Aux analog in 0,No power]

This means that it is not possible to communicate with the MCP3221, normally due to it not beeing powered but could be other issues with the I2C bus.

DBS3262, 2024-10-31:
So Ive tried to eliminate some variables. In my previous post I was trying to take the E5X "maker" board with the Phil Bartlett Isolated ADC to input the arc voltage to IOsender. The E5X was untested by manufacture and they were unable to help specify the appropriate pull-up resistors. This obviously left open the possibility I damaged something in testing. I have since bought and assembled Phil's breakout board (T41U5XBB) and paired it with his isolated ADC.  Here are some pictures of bench test:

This is the recommended setup from Phils manual:
[image]

This is the V_ISO input voltage:
[image]

This is Vin (arc voltage) with pot set mid range:
[image]

Unfortunately, despite using a completely different board, firmware config,  Im having the same problems as before.

1. Plasma Fly out missing.
2. Limited plasma config options: 350, 351, 358, 359, 360.
3. Error msg: Plasma mode is not available" when spindle is engaged.
4. No voltage on MCP3221

[image]
[image]

 To rule out problem with the ADC I tested the it with an Arduino sketch and it appears to be functioning.  The rise and fall of the graph was the manipulation of the potentiometer. 

[image]
 
Here are screen shots from the web builder showing my selections. 

[image]
[image]
[image]
[image]
[image]
[image]

Im not sure what I'm doing wrong. I appreciate your help....

terjeio, 2024-10-31:
I'll look into this when I am back home late next week.

DBS3262, 2024-10-31:
Thank You. I appreciate it!

DBS3262, 2024-11-07:
INPUT voltage at V_ISO_IN: 12.06
OUTPUT voltage at U3: 5.06 VDC

[image]

DBS3262, 2024-11-07:
Yes, I am getting the same result. 

Did you test a freshly flashed board?
FYI:  Ive tried the other configuration ( plasma plug-in w/o MCP3221 I2C ADC) and I get different results. The plasma fly-out is available and I don't see the error codes when the spindle is called. In short appears to work properly. When I re-Flash and include the MCP3221 option, the plasma tab for the fly-out is missing from the Grbl screen.  I  can see the voltage option in $350, but overall the only options available in settings are: $350, $351, $358, $359, and $360. I've been unable to manipulate the settings that are available to induce either the fly-out tab to appear or other functionality to turn on.

terjeio, 2024-11-11:
When the MCP3221 report says _No power_ the plasma plugin will not enable the settings needed for voltage control.
So the primary issue to resolve is why I2C communication with the ADC does not work.
What is the voltage on the SD and SC pins? If not 3.3V then the I2C bus is stuck.

DBS3262, 2024-11-12:
3.3V present at SD and SC. Just as a reminder, I originally attempted to get this plugin to work on a E5X and now a T41U5XBB. Two boards from two different manufacturers,  having the same problem.  
[image]
[image]
[image]

terjeio, 2024-11-12:
> Two boards from two different manufacturers, having the same problem.

The ADC board is possibly bad then? The one I have is working.

DBS3262, 2024-12-03:
Hey Terjeio,

Hope all is well! 

My problem doesn't appear to be in the hardware. I wrote the  sketch pictured below in the Arduino IDE  requesting 2 bytes of data (12 bit resolution) from the isolated ADC. I flashed it onto the Teensy and connected the i2c lines to the breakout board. (T41U5XBB pictured below.) I then set up the test procedure detailed in the ADC manual.  I can vary the voltage from 0-5 VDC and get the corresponding 0- 4096 reading from the ADC. This verifies 100% of the hardware. These are the same boards Ive been attempting to bench test Iosender and the plasma plug-in with. I don't know if something is not compiling correctly, or there is an error in the code, but something else is causing the problem. I have flashed multiple configurations from the web builder on to this hardware and everything has worked except the MCP32221 I2C ADC input.  If there is something else you would like me to try to get this going, please let me know. 

[image]
[image]
[image]
[image]
[image]
[image]

terjeio, 2024-12-03:
I guess we have to ask @phil-barrett to test with a T41U5XBB board since mine is mounted in a rather inaccessible box in a different building.
I have two T41BB5X Pro boards (v2.20 and v2.52) available for testing and the ADC works with both of them - both Web Builder builds and local builds. The I2C pins are the same for all these boards

phil-barrett, 2024-12-03:
I have one on my desk.  It's reading the voltage and scaling it (0.1 to get 100V nominal). This with a V2.09 T41U5XBB and V1.4 ADC.  Build [VER:1.1f.20241120:]

[image]

[image]

[image]

terjeio, 2024-12-03:
Hmm, is the ADC board powered _after_ the controller has booted? If so init fails and it is flagged as having `No power` and ignored.

phil-barrett, 2024-12-03:
If the board is plugged into the I2C header, it will be powered at boot
time.  The one I show in the (poor) photo above is working - reports arc
voltage and varies when I adjust the potentiometer.

terjeio, 2024-12-03:
> If the board is plugged into the I2C header, it will be powered at boot time.

Only if V_ISO is present... If from an external supply that is coming up after the controller the ADC chip is not powered.

phil-barrett, 2024-12-03:
Ah, yes, you are right.  I have that set up.  Oddly enough, V_ISO is 5V but that is sufficient to run the ADC at about 3.8V (from the VReg on the ADC board) and the isolator doesn't really care. I should probably spec V_ISO down to 5V.  Makes for a more convenient set up.

DBS3262, 2024-12-04:
Super frustratingly funny. I went back and looked over our discussion at how many times you asked me if it was powered. It pains me to say more than once.   It never dawned on me that it had to be powered first, so it could initialize.  After reading todays posts I powered V_ISO before powering the teensy and it worked!!!! Thanks gentlemen! Now that I'm finally getting to play with it I have some more questions, but I'll move that over to discussions.

phil-barrett, 2024-12-04:
I can see how it might not be obvious that it needs 2  power connections.  I will update the manual and make it clear.

Glad you are making progress!

cdagneta, 2025-01-01:
Ugh, I regret that I missed this thread all together. I can confirm I ran into this issue as well. Viso needs to be powered before the teensy boots. 

I assumed it was just the isolated power supply I was using being wonky and taking too long to come up to voltage. 

I solved this on both my machines by adding a delay timer relay in line with the Teensy power feed. It disconnects power from the teensy for 2 seconds when the regular 24v power comes up before powering the teensy.