grblHAL/Plugin_plasma #15: Plasma Up/Dwn signals possibly reversed.

DBS3262, 2025-01-08:
Hey Terje,

Having a problem with Plasma up/down mode. 

Configuration:  plasma mode = Up/Dwn.  Originally configured "plasma up" =  3 and "plasma down" =  2. Inverted I/O ports 2,3. THC wired wired with "up" to ST3 and "down" to ST2 on E5X controller. 

Im testing with Proma THC 150 in test mode. This simulates a floating voltage and sends short up and down signals. THC  has torch up and down indicators so the user knows what the intended output should be. 

When testing with the above configuration, controller was going up during down signal form THC and down during Up signal from THC.  I reassigned I/O ports,  Cutter down = 3, Cutter up =2. This corrected controller motor output to match proma THC; However, plasma fly out indicators continue to show reversed output.

On a side note, G65 macros seem to be working great. I couldn't find integration for ohmic probe yet. I wrote a G65 macro for using a relay to accomplish this. Would be happy to post the details if you think that would be helpful to other users.

terjeio, 2025-01-12:
Yes up/down signals are reversed, will fix in the next commit. Until then just reverse wiring.

> Would be happy to post the details if you think that would be helpful to other users.

Tha would be nice, I have not gotten around to make/set up a plasma table yet so any input for the plugin is appreciated. It would be great to get it out of the experimental state.

DBS3262, 2025-01-14:
Great! We have the same goal. Just posted my ohmic probe idea in "show and tell."  I've been testing it a bit and have found something that's not desirable. Ive been trying to dial in a new THC and have been testing on thinner materials. Occasionally the metal moves faster than the THC can react, and the ohmic probe touches the metal.  When this happens the Ohmic probe sets off an Estop. I feel like it would be preferable to ignore the pin while cutting giving the THC a chance to catch up. Is that something that can be done easily on your end? I could turn the probe on and off with another relay, but wanted to get your opinion on the matter.

terjeio, 2025-01-14:
I have refactored the THC code some, and fixed the up/down issue.
Currently the [up/down pins are checked at a 1ms interval](https://github.com/grblHAL/Plugin_plasma/blob/2be37e546aabad9d0b08a971c6b36b2aed1ce5a8/thc.c#L292-L307) and step pulses are thus output at a constant 1 KHz rate when one is asserted. So the only way to speed up Z with the current code is to decrease the microstepping if > 1. The code _could_ be changed to command a move of a certain distance instead, and perhaps at lower frequency? Somehow under PID control like voltage controlled THC? 

> Is that something that can be done easily on your end?

Modifying handling of signals is easy - they can be masked or injected by adding plugin code. I'll have to check why estop is triggered though - probe protection is somehow enabled?

DBS3262, 2025-01-15:
> So the only way to speed up Z with the current code is to decrease the microstepping if > 1.

I think what you're suggesting is manipulating the steps/mm. If so, this could work very well. I'm envisioning  a "Speed" variable in settings that would be a multiplier that defaults to 1.00. At the default setting the Z will use the configured resolution and drive Z at the modal feed rate. (1x resolution) If you change the speed setting to 2.00, the resolution would double (2 x resolution), telling the controller it has to make twice the steps in the same time frame, causing the motor to move twice as fast. 
My old THC does something similar. The Proma SD gets installed between the controller and driver. It lets all the controller signals pass through to the driver, but it can send its own up/down signals directly to the driver as well. They use a "speed setting" to do the same thing were attempting here, but they manipulate the clock frequency to adjust feed rate. Its not something that gets manipulated often. For general cutting you set the speed fast enough to avoid bottoming out. . Slower speed has better blending but if you're cutting something like corrugated metal, you need increase the speed until you can follow the metal.
The major difference in your approach and theirs....speed will be derivative of the modal feed rate. As metal thickness increases modal feed rate will come down. Conversely, their correction feed rate is a essentially a constant. However,  I don't think it will be problem as thicker metal tends to move less, and if needed, you can always increase the speed multiplier.

terjeio, 2025-01-15:
> I think what you're suggesting is manipulating the steps/mm.

Not dynamically - however I would like to know what your settings for the Z-axis are, including the stepper driver microstepping setting.

> The Proma SD gets installed between the controller and driver. It lets all the controller signals pass through to the driver, but it can send its own up/down signals directly to the driver as well.

When it controls the stepper directly does it output a fixed step rate or accelerate/decelerate it?

Some thougths/ramblings:

Ideally it would be best to accelerate/decelerate to/from the rapids rate in up/down mode, acceleration is doable but deceleration could be more problematic since it will inevitably produce some overshoot which may cause the position to oscillate. If inside the deadband of the circuitry providing the up/down signalling then likely not... Who knows how it will behave...

The best option is of course to use voltage control, then the voltage is used to calculate an error distance that is used to generate rapid moves (PID loop controlled). The acceleration setting in combination with the error distance and PID loop parameters will dictate how fast it will track changes.

FYI there are three different metods to inject steps available, and none passes through the regular step generation:

In up/down mode it is currently fixed at a 1KHz rate. This could be increased but without acceleration we risk stalling the motor?

In voltage control mode step generation is by using an [algorithm](https://github.com/grblHAL/core/blob/master/stepper2.c) derived from an article/code by David Austin - this can be run either from timer controlled interrupts (best) or from the foreground process by checking elapsed time (not so good at high step rates since quite a bit of jitter may be introduced). Some drivers supports timer controlled interrupts, some does not.
I wonder if this could be used in up/down mode as well by commanding short moves instead of using a fixed step rate. I can make that an option so it could be tested.

---

I have tweaked the code a bit more and added a new setting, `$674`, for options:

_Virtual ports_ add support for some LinuxCNC M-codes, `M67E3Q<n>` or `M68E3Q<n>` for [velocity reduction](http://linuxcnc.org/docs/2.8/html/plasma/plasmac-user-guide.html#velocity-reduction), `M62P2` et.al. for [THC on/off](http://linuxcnc.org/docs/2.8/html/plasma/plasmac-user-guide.html#thc).  `M62P3` et.al. for [hole cutting](http://linuxcnc.org/docs/2.8/html/plasma/plasmac-user-guide.html#_hole_cutting) is yet to be implemented but claims the auxiliary output related to it.

[image]

_Sync Z position_ will update the Z-position at the end of THC control (when grblHAL changes to IDLE state) so that the core knows where the cutter is. Not sure if this should be an option or not.

[image]

I'll commit this tommorow.

DBS3262, 2025-01-17:
> I would like to know what your settings for the Z-axis are, including the stepper driver microstepping setting.

I'm currently running 1600 Pulse/ Rev on the drivers and 197.799 on the Z travel resolution.

> When it controls the stepper directly does it output a fixed step rate or accelerate/decelerate it?

I wasn't expecting it to, but yes it does accelerate and decelerate.  Increases frequency with each additional volt for about 12 volts and then levels off. If the speed is setting is increased or decreased the whole scale is adjusted up or down.

Set point 77V Reading on THC 78V Speed setting 50%

[image]

Set Point 77  Reading on THC 82V Speed setting 50%

[image]

Set point 77  Reading  on THC 84V Speed Setting 50%

[image]

Set point 77 Reading on THC 78 Speed setting 100%

[image]
   
 The hysteresis setting on the Proma 150 (up/down model) has a range of 2-30 volts. A 2 volt setting  would be (+1 /-1).  It's default setting is 8 (+4/-4.) Here is a graph from the manual showing its basic operation.

[image]

I haven't had a chance to check out the tweaks, but its on my todo list this weekend.