grblHAL/Plugin_plasma #17: THC Thumbnail / Fly-out not appearing in ioSender when MCP3221 I2C ADC option is selected in Web Builder

runningc1, 2025-01-21:
Hello folks, 

I've been getting my feet wet with a custom cnc plasma cutter (4' x 2' cutting area) using NEMA 23s, roughly based on JDs Garage's XL CNC Plasma Table design (https://jdsgarage.bigcartel.com/product/xl-cnc-plasma-table). I went the Teensy route with Phil Barrett's breakout boards (T41U5XBB  and THC ADCI_1). The setup and installation was a cinch, and I've verified the CNC will move around with the ioSender software. I can home the machine, load Gcode files in ioSender, have it move to zero, start the torch, and cut a small rectangle from a file.

However, in ioSender, I see no flyout from the tools on the right side of the UI, and my options are limited in the "GRBL: settings" section of the UI. During cutting, THC does not appear to be active.

My issue is nearly identical to [THC plugin not working properly, showing false "arc ok" among others](https://github.com/grblHAL/Plugin_plasma/issues/10), but I'm careful to power up the ADCI_1 first (12v to V_ISO) from a separate power supply before anything else is powered (as was the ultimate solution to the problem in that thread).
It is interesting to me that I'm not getting the same "[PIN:MCP3221:0,Aux analog in 0,No power]" message as was the symptom in the other discussion, so I don't think it's a powering order issue. When I dump the $pins I get the following for MCP3221 instead:
[PIN:MCP3221:0,Aux analog in 0,E0]

(What is 'E0'? Does it hold the secret to why it's not working?)

I can read the I2C address for the chip on the ADCI_1 board (0x4d). When the ADCI_1 board is removed, the scanner can't find the address anymore. So at the very least the I2C chip is verified accessible.

For voltage testing on the ADCI_1 board, I’m measuring 12V (supplied) on V_ISO, 3.3v on the header power pin, and 3.3v on both header I2C pins. The SMT regulator U3 has 5v on the regulated side.

Here's more relevant info I can think of:
- Using ioSender 2.0.45 (latest)
- Plugin_plasma version: PLASMA v0.16
- With the Web Builder, I've double checked that the Web Builder has both the THC plasma plugin and the MCP3221 I2C ADC option selected.
- I'm using the teensy.exe program to flash the board with the generated hex file.

$i and $pins dumps:

$i
[VER:1.1f.20250120:]
[OPT:VNMSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,ES,REBOOT,SED]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:484K]
[DRIVER:iMXRT1062]
[DRIVER VERSION:250107] 
[DRIVER OPTIONS:USB.1]
[BOARD:T41U5XBB]
[AUX IO:7,3,1,0]
[PLUGIN:PLASMA v0.16]

$pins
[PIN:16,Feed hold]
[PIN:17,Cycle start]
[PIN:14,Emergency stop]
[PIN:15,Probe]
[PIN:20,X limit min]
[PIN:21,Y limit min]
[PIN:22,Z limit min]
[PIN:36,Aux in 0,P0]
[PIN:30,Aux in 1,P1]
[PIN:34,Aux in 2,P2]
[PIN:35,Aux in 3,P3]
[PIN:41,Aux in 4,P4]
[PIN:28,Aux in 5,P5]
[PIN:29,Aux in 6,P6]
[PIN:MCP3221:0,Aux analog in 0,E0]
[PIN:2,X step]
[PIN:4,Y step]
[PIN:6,Z step]
[PIN:3,X dir]
[PIN:5,Y dir]
[PIN:7,Z dir]
[PIN:10,X enable]
[PIN:40,Y enable]
[PIN:39,Z enable]
[PIN:12,Spindle on]
[PIN:11,Spindle direction]
[PIN:13,Spindle PWM]
[PIN:18,Mist]
[PIN:19,Flood]
[PIN:31,Aux out 0,P0]
[PIN:32,Aux out 1,P1]
[PIN:33,Aux out 2,P2]
[PIN:24,I2C SCK]
[PIN:25,I2C SDA]
[PIN:0,RX,UART1]
[PIN:1,TX,UART1]

I'm at a total loss for debugging this thing. Any ideas are welcome.

terjeio, 2025-01-21:
Aux ports has to be configured, and some plasma settings are only available if relevant ports are. `$RST=&` can be used to let the plugin try to set some sensible defaults. If configuration is not good the THC flyout will not show up.

[image]

runningc1, 2025-01-22:
Based on all the hardware connectivity checks and reports in the software, this is looking more and more like a SW issue in my setup than a HW one.

I tried the $rst=& command, and I ended up with identical settings as you (see image), but no flyout (indicative to me that there is an incomplete config, as per your observation). I reflashed the board with a fresh download of the firmware and connected to a fresh install of ioSender, but no dice.

In addition, I played around with the Plasma $350 setting to see if turning it to Off eliminated the requirement to have the 367 Arc Okay port, but also no change. (Ultimately, my setup will use the arc voltage as the okay signal, so I believe I will be using $350=0 by the end.)

[image]

Makes sense that [PIN:MCP3221] is (I believe) binding to pin 0, as the $pins shows: [PIN:MCP3221:**0**,Aux analog in 0,E0]
Or is my thinking in error here?

This concept of binding Aux ports - is it **only** done automatically? Could I try manually binding? Is there a way to do this in the GRBL Settings or using Gcode commands? I looked through the documentation for GRBLHAL and the THC plugin but nothing seemed to jump out at me as a direct means of doing this. (Might have not read thoroughly enough though.)

For good measure, one more dump of $i and $pins:
```
GrblHAL 1.1f ['$' or '$HELP' for help]
[VER:1.1f.20250120:]
[OPT:VNMSL*,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,ES,REBOOT,SED]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:484K]
[DRIVER:iMXRT1062]
[DRIVER VERSION:250107]
[DRIVER OPTIONS:USB.1]
[BOARD:T41U5XBB]
[AUX IO:7,3,1,0]
[PLUGIN:PLASMA v0.16]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
[VER:1.1f.20250120:]
[OPT:VNMSL*,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,ES,REBOOT,SED]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:484K]
[DRIVER:iMXRT1062]
[DRIVER VERSION:250107]
[DRIVER OPTIONS:USB.1]
[BOARD:T41U5XBB]
[AUX IO:7,3,1,0]
[PLUGIN:PLASMA v0.16]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
$pins
[PIN:16,Feed hold]
[PIN:17,Cycle start]
[PIN:14,Emergency stop]
[PIN:15,Probe]
[PIN:20,X limit min]
[PIN:21,Y limit min]
[PIN:22,Z limit min]
[PIN:36,Aux in 0,P0]
[PIN:30,Aux in 1,P1]
[PIN:34,Aux in 2,P2]
[PIN:35,Aux in 3,P3]
[PIN:41,Aux in 4,P4]
[PIN:28,Aux in 5,P5]
[PIN:29,Aux in 6,P6]
[PIN:MCP3221:0,Aux analog in 0,E0]
[PIN:2,X step]
[PIN:4,Y step]
[PIN:6,Z step]
[PIN:3,X dir]
[PIN:5,Y dir]
[PIN:7,Z dir]
[PIN:10,X enable]
[PIN:40,Y enable]
[PIN:39,Z enable]
[PIN:12,Spindle on]
[PIN:11,Spindle direction]
[PIN:13,Spindle PWM]
[PIN:18,Mist]
[PIN:19,Flood]
[PIN:31,Aux out 0,P0]
[PIN:32,Aux out 1,P1]
[PIN:33,Aux out 2,P2]
[PIN:24,I2C SCK]
[PIN:25,I2C SDA]
[PIN:0,RX,UART1]
[PIN:1,TX,UART1]
```

Thanks again for your help.

terjeio, 2025-01-22:
> (What is 'E0'? Does it hold the secret to why it's not working?)

Kind of, the iMXRT1062 driver misbehaved with the latest plugin leading to the pin not beeing claimed, fixed now. 

> This concept of binding Aux ports - is it only done automatically?

No, only on a settings reset and on first startup after adding the plugin.

> Could I try manually binding?

Yes, binding ports is in the _Aux ports_ group if using ioSender.

> Is there a way to do this in the GRBL Settings or using Gcode commands? 

Both are possible. I have to add the setting numbers to the documentation, earlier it was only done automatically and I am sloppy with keeping the documentation in sync...