grblHAL/Plugin_plasma #18: Warning: Plasma mode not available!

4MoH9I-HUB, 2025-01-31:
I have such a problem on the microcontroller STM32F401.
I use the driver STM32F4XX in CubeIDE. The project is built without errors and warnings. If I take a new board, or completely erase the old one, then the Plasma initializes and works. But as soon as I press Reset on it or turn off the power, IOSender gives an error
How can I solve this problem, where should I look for a solution?

4MoH9I-HUB, 2025-01-31:
Here is the project I use

[STM32F4xx-master.zip](https://github.com/user-attachments/files/18615489/STM32F4xx-master.zip)

terjeio, 2025-02-01:
I did not pay attention to the issue title... For THC to be functional both cutter up and down has to be assigned to ports, cutter up is not defined in your configuration. Set the port to 2 with `$369=2`.
FYI in the next commit I have started implementing a "plain" mode that only needs the arc ok input - it has no THC control, only arc on handling and arc ok monitoring.

closed 2025-02-19 by 4MoH9I-HUB