grblHAL/Plugin_plasma #20: Ethernet-UART

4MoH9I-HUB, 2025-02-19:
I have an interface converter module. I tried to connect it directly to my controller, configured for UART, but no matter how hard I tried, I couldn't do it, although I'm sure that the module itself is configured correctly. Could this be due to the fact that connecting via the ehternet requires its initialization in the controller, and simple interface conversion will not help here?

converter:
https://www.pusr.com/products/serial-to-ethernet-converter-modules-usr-tcp232-t2.html

4MoH9I-HUB, 2025-02-19:
I forgot to say, I use a driver STM32F4

terjeio, 2025-02-19:
Have you set set up the converter module correctly? Easiest is likely via its web interface.

4MoH9I-HUB, 2025-02-24:
This module is not in GrblHal. 
So I ask, is it possible to connect through this module, setting up the controller on UART?

terjeio, 2025-02-24:
> So I ask, is it possible to connect through this module, setting up the controller on UART?

It should, you have to build the controller firmware with UART as the default communication channel. You are compiling with the Web Builder?
The module has to be preconfigured via its Web interface or a [plugin like the ESP-AT](https://github.com/grblHAL/Plugins_misc/?tab=readme-ov-file#esp-at) has to be written for it.

4MoH9I-HUB, 2025-02-25:
everything is correct. I compiled the project using CubeIDE and configured the project just for UART and it works fine. But when I connect my module directly and try to connect via IOSender in Ethernet nothing happens. To connect I use the IP and port settings of my module

4MoH9I-HUB, 2025-02-25:
as far as I understand, to work with this module, you need a plugin like for W5500?
By the way, I also have W5500 available and I tried to establish a connection through it too. However, this causes some difficulties.

terjeio, 2025-02-25:
> as far as I understand, to work with this module, you need a plugin like for W5500?

No, it is a bridge to UART - if set up correctly via its web interface it _should_ work with _no_ changes to a controller set up for UART communication. If it does not work it could be incorrect wiring or baud rate or what not. But you have provided no details about which board, how you have wired (a photo may help) or other stuff that could be useful for me trying to understand what you are doing. A plugin like the ESP-AT linked above _could_ be useful but is not needed since the web interface should provide you with the neccesary options to set it up. BTW are you able to connect to it?

> By the way, I also have W5500 available and I tried to establish a connection through it too. However, this causes some difficulties.

The W5500 uses SPI comms and indeed is a bit complicated to set up - if at all possible with the board you have.

4MoH9I-HUB, 2025-02-25:
I will tell you in detail. I have a CNC controller board. Communication with the computer occurs through this circuit: 
USB-CH340-MAX3232-UART~~~~~~UART-MAX3232-TTL-STM32F401. 
This circuit has been tested and works fine. However, now the task is to use the Ethernet interface instead of the UART interface. For this, I want to use the TSP232-T2 module (https://www.pusr.com/products/serial-to-ethernet-converter-modules-usr-tcp232-t2.html). The connection diagram in this case: 
EZERNET(PC)~~~~~~~module T2-MAX3232-STM32F401. 
I configured the module and checked it through the UART LOOP mode. 

[image]

[image]

The data that it sends, it receives. I also checked through the console by entering the ping <IP> command. Data packets were also returned without loss, regardless of whether the UART was connected or not. 

[image]

However, when I try to connect to my board, nothing happens. (connection could not be established)

[image]

I checked and tried different ways to connect UART. I tried to connect it via MAX3232 and directly to the controller. I swapped RX and TX in both cases. Searching for a problem in the module itself did not lead me to success, since regardless of what type the module was configured for (CLEENT/SERVER  TCP/UPD), the result did not change. That is why I had the feeling that this module cannot be launched without an additional plugin.

terjeio, 2025-03-01:
If you are able to connect a serial terminal to the UART side of the module and data is passed unmodified trough it then it can be used. I do most of my development via a UART connection (UART <> USB bridges) since I do not lose the connection when I reflash so I know UART comms is working.

BTW which board? If a Nucleo-64 you cannot connect an external UART to the default pins since these are wired to the debug probe which provides a UART <> USB bridge.