grblHAL/Plugin_plasma #21: Where is the Target Voltage ?

Francois-le-francais-dev, 2025-02-21:
How can I define the "target voltage" for GRBLHAL THC please ?
Is there a GCODE ?
Is there a special $xxx command ?
I can't find anything about it.

I try to use IOSender with a T41U5XBB board with ADCI from Ph. Barrett on a DIY plasma cutter.

Many thanks !

terjeio, 2025-02-22:
It is [set automatically](https://github.com/grblHAL/Plugin_plasma/blob/a560035fb8dc1a6bf6a9004bb5d32b52bba54161/thc.c#L415). If there is a need to set it manually I have to check the LinuxCNC spec for how they do it - I cannot remember seeing any M-code for setting it.

Francois-le-francais-dev, 2025-02-22:
In LinuxCNC QtPlasmaC the target voltage is set in the material spec (§9.6 Material File) : https://linuxcnc.org/docs/html/plasma/qtplasmac.html#plasma:material-handling.
It is a way to include it in the GCODE by a "Magic comment" (see §9.9 ).

terjeio, 2025-02-22:
Note that ioSender (and possibly others) strip comments by default. I have just uploaded a new [edge version](https://github.com/terjeio/ioSender#edge-pre-releases) that sends them if configured.

Saving materials to a file is currently not supported, but you should be able to upload them for the duration of the session making them available for selection via M190. Only arc voltage and pierce delay is used for now as I am unsure about how to use the other ones. All parameters are currently parsed and there is a check for presence of the mandatory ones Also, the parser is quite restrictive so use the format in the example provided via the link above:

`(o=0, nu=2, na=5mm Mild Steel 40A, ph=3.1, pd=0.1, ch=0.75, fr=3000, kw=0.5, th=1, ca=45, cv=110, pe=0.1, gp=5, cm=1, jh=0, jd=0)`

Non-mandatory parameters can be omitted and the order is not important except the `o` parameter has to be first.

I do not have a plasma cutter to test with so I do not know if this will work as-is, likely some tuning will be required.

Francois-le-francais-dev, 2025-02-23:
Than you for the quick add.
I tested 2.0.46p5 but magic comment appears to have no effect.
THC is working like before, cut voltage param "cv" is ignored.
It seems M190 Pn code is not supported, I have "error:20" and "unsupported or invalid gcode command".

What is the good way to insert the magic comment in the gcode file, is there any rule ?
What about M190 command : is it added by THC plugin ?

I can't go further by now.

terjeio, 2025-02-23:
The controller firmware has to be updated- and I am committing [new functionality](https://github.com/grblHAL/Plugin_plasma#materials) while writing this.

> What is the good way to insert the magic comment in the gcode file, is there any rule ?

Only that the comment starts with `(o=` and contains the mandatory elements. An example that sets parameters effective immediately:

`(o=0, na=grblHAL test, ph=3.1, pd=0.1, ch=0.75, fr=310, th=1, cv=103)`

> What about M190 command : is it added by THC plugin ?

Yes, and parsing of magic comments as well. Magic comments are ignored if the THC plugin is not installed. And if a material number as specified by the M190 P-word is not loaded an error will be returned.

New in the commit is that pierce height (ph) and cut height (ch) is acted on now.

The updated plugin reports itself with v0.20:
[image]

Francois-le-francais-dev, 2025-02-25:
I have to stop for a while because I burnt my ADCI (a stupid wiring mistake).
I have no money for buying a new one for now (and shipping to France is expensive).
I would need to have more than one because Im not an electronics engineer and I work alone ! ;o)

I will let you know when I'm ready to test again.
I don't give up, I have still a lot of questions and suggestions to make it clear and work fine, for example : 

1. what is the number of the material created by the THC plugin ? where to find it ?
2. better : could you explain step by step (with an example and screens maybe) how to load a file with a material in a magic comment and to manage materials in IOSender ? I didn't succeed to that.
3. is it possible to insert the THC popup definitively in the main window (if loaded) ?
4. it would be fine to see the cut voltage and others material param in a clear persistant way too,
5. general : when I cut sheets I often have more than a hundred of parts in a single sheet and Im used to select a part graphically in the "3D" window because it's very difficult to find it in the code (plasma cutting = often stop and resume where you are) : could it be possible to group code lines by shapes and select these shapes in the graphical window ? I created a postprocessor for SheetCAM that makes that kind of job for cutting with bCNC. I can make one for IOSender too.

Thanks for the good progress you gave to the project so quickly ! 👍

terjeio, 2025-03-01:
1. List the materials with the `$ES` command, it is the number following  `nu=`. When I get time I will add a list populated from the `$ES` output to ioSender making all the details available in an easy to read format. And click to set a selected material as default in the controller via `M190`.
2. Add lines like this `(o=0, na=grblHAL test, ph=3.1, pd=0.1, ch=0.75, fr=310, th=1, cv=103)` to your gcode. If this is not working check that the plasma plugin is installed and ensure it is the latest version. And use the `$EM` command to list the already stored ones.
3. Yes, but not without changing code.
4. Code has to be added for this, at least in the last version a message will be shown in the status line when a new material is selected.
5. Again, code is needed for that - and is not trivial since shapes has to be extracted from the gcode. Or CAM capability added to ioSender...